mandulaj/PZEM-004T-v30 #104: I can not read 3 pzem

sanluca, 2023-02-19:
Hi all, 
sorry if I write here but I can't understand what's going on, before I had two pzems and everything worked fine, now I added the third one, but something strange happens, the pzems seem to be read but the result is all written on the third pzem, but when I go to read the addresses are correct, I hope I explained myself well, I attach the code thanks

`//************************************************************
//
// Read Consuption with Pzem-004
// wemos d1 mini pro
//
//************************************************************
float voltg,curg,powerg,energ,freqg,pfg,volta,cura,powera,enera,freqa,pfa,voltc,curc,powerc,enerc,freqc,pfc;
String addr,addr2,addr3;

unsigned long thisMillis = 0;
unsigned long lastMillis = 0;
unsigned long interval = 300000;//5 min
unsigned long now;
int count=0;

#include <PZEM004Tv30.h>
//PZEM004Tv30 pzem(&Serial3);
PZEM004Tv30 pzem (D5, D6, 0x43);
PZEM004Tv30 pzem2 (D5, D6, 0x42);
PZEM004Tv30 pzem3 (D5, D6, 0x44);
//#include "painlessMesh.h"
#include "namedMesh.h"

#define   MESH_PREFIX     "whateverYouLike"
#define   MESH_PASSWORD   "somethingSneaky"
#define   MESH_PORT       5555

Scheduler userScheduler; // to control your personal task
//painlessMesh  mesh;
namedMesh  mesh;
String msg;
String rel;
String rel2;
String nodeName = "enel"; // Name needs to be unique
String to = "bridgemqtt";
uint32_t root_id;

#define ROLE    "enel"
#define VERSION "Enel v2.0.7"
#define MESSAGE "enel "

// User stub
void sendMessage() ; // Prototype so PlatformIO doesn't complain

Task taskSendMessage( TASK_SECOND * 1 , TASK_FOREVER, &sendMessage );

void sendMessage() {
  //generale
  read_enel_gen();
//appartamento
  read_enel_ap();
//cantina garage
  read_enel_cant();

  taskSendMessage.setInterval( random( TASK_SECOND * 3, TASK_SECOND * 5 ));

}

// Needed for painless library
void receivedCallback( uint32_t from, String &msg ) {
  Serial.printf("startHere: Received from %u msg=%s\n", from, msg.c_str());
  rel=msg;
  root_id=from;
}

void update_reset(){
  if (rel2 != rel){
  if (strcmp(rel.c_str(),"1") == 0){
    pzem.resetEnergy();
    pzem2.resetEnergy();
    pzem3.resetEnergy();
}
rel2=rel;
}}

void newConnectionCallback(uint32_t nodeId) {
    Serial.printf("--> startHere: New Connection, nodeId = %u\n", nodeId);
}

void changedConnectionCallback() {
  Serial.printf("Changed connections\n");
}

void nodeTimeAdjustedCallback(int32_t offset) {
    Serial.printf("Adjusted time %u. Offset = %d\n", mesh.getNodeTime(),offset);
}

void read_enel_gen() {

  voltg = pzem.voltage();
  curg = pzem.current();
  powerg = pzem.power();
  energ = pzem.energy();
  freqg = pzem.frequency();
  pfg = pzem.pf();
  addr = pzem.getAddress(), HEX;
  msg = "voltg/";
  msg += voltg;
  mesh.sendSingle(to, msg);
  msg = "curg/";
  msg += curg;
  mesh.sendSingle(to, msg);
  msg = "powerg/";
  msg += powerg;
  mesh.sendSingle(to, msg);
  msg = "energ/";
  msg += energ;
  mesh.sendSingle(to, msg);
  msg = "freqg/";
  msg += freqg;
  mesh.sendSingle(to, msg);
  msg = "pfg/";
  msg += pfg;
  mesh.sendSingle(to, msg);
  msg = "addr/";
  msg += addr;
  mesh.sendSingle(to, msg);
}

void read_enel_ap() {

  volta = pzem2.voltage();
  cura = pzem2.current();
  powera = pzem2.power();
  enera = pzem2.energy();
  freqa = pzem2.frequency();
  pfa = pzem2.pf();
  addr2 = pzem2.getAddress(), HEX;
  msg = "volta/";
  msg += volta;
  mesh.sendSingle(to, msg);
  msg = "cura/";
  msg += cura;
  mesh.sendSingle(to, msg);
  msg = "powera/";
  msg += powera;
  mesh.sendSingle(to, msg);
  msg = "enera/";
  msg += enera;
  mesh.sendSingle(to, msg);
  msg = "freqa/";
  msg += freqa;
  mesh.sendSingle(to, msg);
  msg = "pfa/";
  msg += pfa;
  mesh.sendSingle(to, msg);
  msg = "addr2/";
  msg += addr2;
  mesh.sendSingle(to, msg);

}

void read_enel_cant() {

  voltc = pzem3.voltage();
  curc = pzem3.current();
  powerc = pzem3.power();
  enerc = pzem3.energy();
  freqc = pzem3.frequency();
  pfc = pzem3.pf();
  addr3 = pzem3.getAddress(), HEX;
  msg = "voltc/";
  msg += voltc;
  mesh.sendSingle(to, msg);
  msg = "curc/";
  msg += curc;
  mesh.sendSingle(to, msg);
  msg = "powerc/";
  msg += powerc;
  mesh.sendSingle(to, msg);
  msg = "enerc/";
  msg += enerc;
  mesh.sendSingle(to, msg);
  msg = "freqc/";
  msg += freqc;
  mesh.sendSingle(to, msg);
  msg = "pfc/";
  msg += pfc;
  mesh.sendSingle(to, msg);
  msg = "addr3/";
  msg += addr3;
  mesh.sendSingle(to, msg);
}

void update_status()
{
  thisMillis = millis();
  if(thisMillis - lastMillis > interval){
  long uptime = millis() / 60000L;
  msg = "uptime/";
  msg += uptime;
  mesh.sendSingle(to, msg);
  msg = "nodeid/";
  msg += mesh.getNodeId();
  mesh.sendSingle(to, msg);
  msg = "freememory/";
  msg += String(ESP.getFreeHeap());
  mesh.sendSingle(to, msg);
  msg = "version/";
  msg += VERSION;
  mesh.sendSingle(to, msg);
  msg = "root/";
  msg += root_id;
  mesh.sendSingle(to, msg);
  msg = "ip/";
  msg += WiFi.localIP().toString();
  mesh.sendSingle(to, msg);
  
  lastMillis= thisMillis;
}}

void setup() {
  Serial.begin(115200);

//mesh.setDebugMsgTypes( ERROR | MESH_STATUS | CONNECTION | SYNC | COMMUNICATION | GENERAL | MSG_TYPES | REMOTE ); // all types on
  mesh.setDebugMsgTypes( ERROR | STARTUP );  // set before init() so that you can see startup messages

  mesh.init( MESH_PREFIX, MESH_PASSWORD, &userScheduler, MESH_PORT, WIFI_AP_STA, 11 );
  mesh.initOTAReceive(ROLE);
  mesh.setContainsRoot(true);
  mesh.setName(nodeName);
  mesh.onReceive(&receivedCallback);
  mesh.onNewConnection(&newConnectionCallback);
  mesh.onChangedConnections(&changedConnectionCallback);
  mesh.onNodeTimeAdjusted(&nodeTimeAdjustedCallback);
  

  userScheduler.addTask( taskSendMessage );
  taskSendMessage.enable();
}

void loop() {
  // it will run the user scheduler as well
  mesh.update();
  update_reset(); 
  update_status();
 if((millis() - now) > 600000) { //600000 - 10 min
    now=millis();
      if(!(mesh.isConnected(root_id))){
        Serial.printf("Reboot\n");
        ESP.restart();
    }}
  }
`

mandulaj, 2023-02-20:
Hi @sanluca, I am sorry, I don't follow what your problem is. What do you mean "result is all written on the third pzem"  You mean, you only get results from the 3rd PZEM? 

Have you set the Addresses correctly using the [PZEMChangeAddress](https://github.com/mandulaj/PZEM-004T-v30/tree/master/examples/PZEMChangeAddress) script? 

Could you maybe also share an example from the output that illustrates your problem to help with the debugging?

I don't really see anything wrong with the code reading the PZEM values. So some output from the messages that are sent would be more useful.

sanluca, 2023-02-20:
I don't know what happened but they lost their addresses, I did various tests but I think something happened to the pzems, so I connected the 3 pzems to 3 different serials and now everything works, thanks

mandulaj, 2023-02-21:
Ok, good that everything was resolved. Unless you have any other problems, I am going to close this issue.

closed 2023-02-21 by mandulaj