mandulaj/PZEM-004T-v30 #105: Pzem-004t v3 + Esp32 DEVKITVQ

M4carloz, 2023-02-27:


mandulaj, 2023-02-28:
Hi @M4carloz, I will reply in English. What happens if you power the PZEM with 3.3V and connect the RX/TX lines directly from the ESP32. (Without using any voltage divider).

You see, the 5V power only powers the opto-coupler, not the actual PZEM chip. Also please try swapping the RX/TX lines, just in case you have them the wrong way around. 

I am not sure which sketch you are using, but if you use this sketch [PZEMHardSerial](https://github.com/mandulaj/PZEM-004T-v30/blob/master/examples/PZEMHardSerial/PZEMHardSerial.cpp) it will communicate using the default (broadcast) address, so you shouldn't be getting any Errors setting an address.

M4carloz, 2023-02-28:
Muito obrigado irmão, funcionou perfeitamente.

[image]

[image]

closed 2023-02-28 by M4carloz