mandulaj/PZEM-004T-v30 #108: when I try to upload, I get an error like this: no matching function for call to 'PZEM004Tv30::PZEM004Tv30(int, int)

mandulaj, 2023-03-22:
when I try to upload, I get an error like this: no matching function for call to 'PZEM004Tv30::PZEM004Tv30(int, int)

_Originally posted by @AwanDami in https://github.com/mandulaj/PZEM-004T-v30/issues/63#issuecomment-1479828573_

mandulaj, 2023-03-22:
@AwanDami Could you share the code you are using?  Also have you installed the library correctly according to the instructions? 

It sounds like the compiler can't find some of the library files.

AwanDami, 2023-03-23:
I use the code to change the address to pzem 004t but there is an error

`#include <PZEM004Tv30.h>

PZEM004Tv30 pzem(5,4);

void setup() {
  Serial.begin(115200);
}

uint8_t addr = 0x07; //THIS IS THE ADDRESS

void loop() {
    pzem.setAddress(addr);
    Serial.print("Current address:");
    Serial.println(pzem.getAddress());
    Serial.println();
    delay(1000);
`

mandulaj, 2023-04-25:
Can you verify that you have the library installed correctly according to the instruction?

ghtfujita, 2023-07-18:
Hi. I have exactly the same issue with exactly the same code.
I have tried several methods to install libraries (from Arduino IDE library manager and from zip), but neither methods worked.
If I delete parentheses and pin numbers(i.e. `PZEM004Tv30 pzem;` ) it succeeds compiling, but in this case , I cannot be sure where RX and TX pins are set to be.
Any suggestions so that I can set slave address to my PZEM?

mandulaj, 2023-07-19:
@ghtfujita which board are you compiling for?

ghtfujita, 2023-07-19:
Hello, @mandulaj . Thanks for your reply.
The board I tried is ESP32-DEVKIT-C. The error message said exactly whats in the thread subject, but I figured out the way to avoid this error.

Since when you google "PZEM-004T Address", its likely you get [this](https://github.com/mandulaj/PZEM-004T-v30/issues/63) page close to the top which does not work, it might be good to have how-to set address section in README.

For others getting the same errors like me, below is the code worked for my environment:
```
#include <PZEM004Tv30.h>
#define PZEM_RX_PIN 25
#define PZEM_TX_PIN 26
#define PZEM_SERIAL Serial2
PZEM004Tv30 pzem(PZEM_SERIAL, PZEM_RX_PIN, PZEM_TX_PIN);

#define SET_ADDRESS 0xA2

void setup() {
    Serial.begin(115200);
    delay(1000);
    Serial.print("old address: ");
    Serial.println(pzem.readAddress(), HEX);
    Serial.print("new address:");
    pzem.setAddress(SET_ADDRESS);
    Serial.println(pzem.readAddress(), HEX);
}

void loop() {
}
```

since the guy originally posted the question seems no more responding, I suppose we can close this thread.

mandulaj, 2023-07-20:
Yes, that is the issue. The constructor `PZEM004Tv30 pzem(PZEM_RX_PIN, PZEM_TX_PIN);` is for using Software serial. 
**Software Serial is not available on the ESP32 platform** 
Thus you get an error. 

You have to explicitly pass the hardware serial with `PZEM004Tv30 pzem(PZEM_SERIAL, PZEM_RX_PIN, PZEM_TX_PIN);` as you have done.

I will close the issue.

closed 2023-07-20 by mandulaj