mandulaj/PZEM-004T-v30 #113: Voltage measurement up to 500 volts

amir-eshaqy, 2023-07-30:
Hello,
   I managed to measure the voltage of 500 volts with the resistance value of R17 to 1.8 megaohm without error in the frequency, but I had a problem in setting the voltage multiplier to show the correct result on the display. The voltage conversion factor is 1333.33 in normal mode but with change The resistance value of R17 is 1.8 megaohm, this value is changed to 2400. How can I apply this coefficient change in the program?

mandulaj, 2023-08-01:
Hey,
The PZEM chip already returns the converted voltage directly in the response message. If you change the resistance value, you will have to manually scale the voltage by the appropriate coefficient. You could do something like this:
```cpp
const float VOLTAGE_SCALE_FACTOR = <the scale factor>;
float voltage = pzem.voltage() * VOLTAGE_SCALE_FACTOR;
```
From what you say it sounds like the conversion factor should be 2400/1333 = 1.8 but you will have to double check that.

amir-eshaqy, 2023-08-02:
> Hey, The PZEM chip already returns the converted voltage directly in the response message. If you change the resistance value, you will have to manually scale the voltage by the appropriate coefficient. You could do something like this:
> 
> ```c++
> const float VOLTAGE_SCALE_FACTOR = <the scale factor>;
> float voltage = pzem.voltage() * VOLTAGE_SCALE_FACTOR;
> ```
> 
> From what you say it sounds like the conversion factor should be 2400/1333 = 1.8 but you will have to double check that.

Thank you, I will definitely check

closed 2024-03-26 by mandulaj