mandulaj/PZEM-004T-v30 #114: Pzem measuring wrong voltage values

Noor-H, 2023-09-15:
Hi,
I am using 4 pzems (004t-100A) on SERIAL1 (RX 26, TX 27) and 4 pzems on SERIAL 2 (RX 16, TX 17)of ESP32 DEV Module. All pzems have different addresses that I have modified and tested individually.
 I am measuring voltage, current and some other parameters of an electrical system. Everything was working fine since a few months. However a few days ago, voltage is showing incorrect values, for example if the original values are between 220 to 240 V, it shows values like 189, 270 etc. which are way off from the original values. There are no voltage surges or spikes. Current values are still correct. The whole system is in a well cooled area and there is no problem of components heating up.
Can you please point out any problem as of why it suddenly started to happen, it will be very helpful.

mandulaj, 2023-09-15:
Strange, never had that happen. Can you verify the voltage with a multimeter? Also have you tried resetting the chip by just unplugging (from AC) and connecting it back?

Noor-H, 2023-09-19:
I did reset it and verify the voltage. The voltage is between 220-240 at some points and 270 at the wire going in the pzem. I think it is due to high flux density probably, however, I am trying to change the wire and connectors to see if it improves the condition.

closed 2023-09-20 by Noor-H