mandulaj/PZEM-004T-v30 #117: How to modify this code to only output certain values

ham3dt, 2023-11-30:
I only need voltage, current and power factor, How can I modify this library to not request/receive unwanted values such as power etc to save time and resources?

mandulaj, 2023-11-30:
Check out the `updateValues` [function](https://github.com/mandulaj/PZEM-004T-v30/blob/eec09ab4574db567306a3d335e5131c766e3333e/src/PZEM004Tv30.cpp#L397).

There is a request to grab 10 registers starting at `0x00`:
```cpp
    // Read 10 registers starting at 0x00 (no check)
    sendCmd8(CMD_RIR, 0x00, 0x0A, false);
```
The register addresses for different registers are:
```cpp
#define REG_VOLTAGE     0x0000
#define REG_CURRENT_L   0x0001
#define REG_CURRENT_H   0X0002
#define REG_POWER_L     0x0003
#define REG_POWER_H     0x0004
#define REG_ENERGY_L    0x0005
#define REG_ENERGY_H    0x0006
#define REG_FREQUENCY   0x0007
#define REG_PF          0x0008
#define REG_ALARM       0x0009
```
So you could use that to read only a specific register and decode it yourself. However I recommend reading as may registers in one go as possible as there will be some checksums that are appended to the end of the message and you want to avoid as much of the overhead as possible. 

Originally there was the idea to implement this functionality, nevertheless, at the end you don't get much by just reading 3/10 registers.

ham3dt, 2023-11-30:
Thanks, @mandulaj 
"Originally there was the idea to implement this functionality", Why did you dropped it then?
This is the only -proper- library for the device, there's so much potential.

mandulaj, 2023-11-30:
As I said, there was little benefit in reading just select registers due to all the protocol overhead:

## Example:
Reading just current:
**You send** 8 bytes
**You receive** 9 bytes
= 17 bytes transmitted = 17 bytes / value

Reading all 10 registers is as follows:
**You send** 8 bytes
**You receive** 25 bytes
= 33 bytes transmitted  = 5 bytes / value

Since for your case you need to read the voltage current and power factor, You could do this: read voltage + current together (as they are consecutive)  and then the PF

**You send** 8 bytes
**You receive** 11 bytes
**You send** 8 bytes
**You receive** 7 bytes
= 34 bytes transmitted......

You see what I mean?

closed 2024-03-26 by mandulaj