mandulaj/PZEM-004T-v30 #119: Check if module is connected

pvoncken, 2024-02-18:
I'm using this library to monitor my solar panels.
Works perfect 
But now i noticed that in case the fuse of that group blows my arduino doesn't update the sensors anymore.
I undserstand that this is caused because the PZEM's need 230Volt to work.

It would be be nice if it was possible to check if a PZEM-adress is still connected.

mandulaj, 2024-02-19:
You will be getting Errors returned from the Sensor (values are Nan). Perhaps you can just perform a check on that and if you get errors, you raise the alarm?
```c
if(isnan(voltage)){
        Serial.println("Sensor not connected");
}
 ```

pvoncken, 2024-02-21:
Why didn't I think of that. 
This works perfect.
Thanxs!

closed 2024-02-21 by pvoncken