mandulaj/PZEM-004T-v30 #120: Is it possible to send one command to measure to all devices connected instead of sending multiple separated commands?

Zokkoman, 2024-03-07:
As the number of PZEM modules increase, the time I have to wait for all of their responses also increses since a new command has to be send for each individual reading. Is it possible to send one command to all modules so they start measuring at the same time? Each response would have to be requested individually, but that would take much less time than the measurement itself, I guess.

vortigont, 2024-03-08:
PZEM's are continuously measuring all parameters, not on command request, so it won't make any difference.

mandulaj, 2024-03-11:
Exactly as @vortigont said, the sensors measure continuously in background and sending the `read` command only samples the current values from the registers. If you use all the PZEMs on one bus, there is no way to sample all at the same time as you can't have multiple parallel interactions on one bus. 

The only solution would be to split the PZEMs to different physical buses. However again, in terms of energy, this would not make much difference as the sensors are anyways keeping their own state in the background. For voltage and current I  would anyways not trust the synchronization that you get the measurement values at the  precise  instant of sending the command (due to averaging etc going on in the background) 

So in conclusion I would not worry about it too much.