mandulaj/PZEM-004T-v30 #122: Inconsisten Byte order in CRC check sum

mandulaj, 2024-03-27:
Im using PZEM004t-100A(V3). Im using RX (3) and TX (1). Connected to an AC load and all wires are connected properly. Even the readAddress() is not functioning. I saw in the data sheet that the command format of the master in measuring is Slave Address + 0x04 + Register Address High Byte + Register Address Low Byte + Number
of Registers High Byte + Number of Registers Low Byte + CRC Check High Byte + CRC Check
Low Byte. However in the library, I saw that the CRC low byte came first and CRC high byte second. does this have anything to do with it I cant read and set the address of my pzem from 0xF8 to 0xF7?
It also says that the PZEM has a range of 0x00 to 0xF7 only.

_Originally posted by @Jemeeeeer in https://github.com/mandulaj/PZEM-004T-v30/issues/26#issuecomment-2020583045_

mandulaj, 2024-03-27:
I have to have a closer look at this, as this is something I wrote 2 years ago and don't remember the details anymore. However I remember that there was some error-checking functionality that was supposed to be implemented, however never got around to doing it. So Thanks for bringing it up.

Jemeeeeer, 2024-04-02:
can u try to fix/implement the error checking functionality?

mandulaj, 2024-04-02:
Yes! This is a high priority now. I just need to make sure that the library doesn't break for existing projects so need to run some checks.

Jemeeeeer, 2024-04-04:
Hello. I would like to follow up on this. Any updates? What seems to be the problem?

Jemeeeeer, 2024-04-14:
Hello. Is there an update?

mandulaj, 2024-04-15:
Hi, I am sorry, I am on my vacation and didn't have any time to look at this. Sorry for the inconvenience