mandulaj/PZEM-004T-v30 #123: Lamp cannot dim by TRIAC when connect with PZEM 004T

Kosaki01, 2024-05-09:
I have a triac-dimmer circuit like below with load is incandescent lamp when i connect the N and L to 2 terminals of lamp, they cannot dim and work at max power - maximum brightness althought i just made they dim about 50% of max brightness. 
[image]

How can i fix this? Pls give me some advice for this. Thank you so much.

mandulaj, 2024-05-13:
Are you attaching the PZEM after the triac? If so, please attach it before the triac! The measurement will be the same.

Kosaki01, 2024-05-15:
[image]
Hi, you mean this right?

mandulaj, 2024-05-16:
LOAD is your lamp right, and the Power Triac is connected in series with the AC Line.... Can you attach the PZEM here where the power comes in?
[image]

mandulaj, 2024-05-16:
Ahh, but then if you are interested in the voltage, PZEM is not the right sensor to use! PZEM is good for measuring the power/energy! The AC input is a continuous sine wave, but the triac will chop it up (which is how it achieves the power modulation)

Kosaki01, 2024-05-16:
Yeah the main reason i used pzem is for measuring power, but with the fault of voltage, they cannot give right value of power :(

mandulaj, 2024-05-16:
But if you do as I told you and attach the PZEM before the Triac and put the current probe around one of the wires, the power measurement should be done correctly. The PZEM will take the correct integrated product of voltage and current over time.

Kosaki01, 2024-05-16:
[image]
Yeah i did it, and i compared pzem's value with VOM's value, they had difference power and voltage value when lamp is dimmed, and same value when lamp is at max brighness - max power comsumption.

mandulaj, 2024-05-17:
With your multimeter, did you measure voltage and current independently? Because if yes, their product is not the actual power for a AC system. A dimmed light will not have a power factor of 1. The Voltage and Current are not in phase! Thus if you only look at the magnitude of the Voltage and Current and their product, you won't get the Real power (There is some reactive power) https://en.wikipedia.org/wiki/AC_power

Kosaki01, 2024-05-19:
So if i measure with an oscilloscope, i will get the real power, right?