mandulaj/PZEM-004T-v30 #124: More than 4 pzem-004t

micheldufresne, 2024-05-19:
Hello
I bought 3 pzem-004t 10A and then 3 others of 100A. I gave the addresses 1, 2, 3, 4 to the first but impossible to give the addresses 5 and 6. When I give an address >4 the pzem takes the address 1.
I use this code:
void initpzem(PZEM004Tv30 pz,uint8_t addr)
{
     pz.setAddress(addr);
     delay(1000);
     Serial.print("Current address:");
     Serial.println(pz.getAddress());
}
Is this normal? Is it impossible to use 6 pzem?
THANKS

mandulaj, 2024-05-28:
When you assign the addresses, are you making sure to only have the one PZEM connected on the bus and attached to the AC power supply at a time?

micheldufresne, 2024-05-28:
Hello,
Yes I am sure.
Il solve my problem using IDE arduino and this code :

/*
Origin Copyright (c) 2021 Jakub Mandula
Copyright (c) 2022 Cyril Poissonnier

PZEM modules adress changer .
================================================================

Each PZEM modules has two addresses. A default general address (0xF8),
which every module will listen on by default. And a custom address.

The custom address can be used if multiple PZEM modules are to be used
on a single ModBUS.

This script will use the general address in order to set the custom
address of the connected PZEM module.

Therefore make sure only one PZEM module is connected at a time!
Otherwise all connected modules will receive the same custom address.

the address can be change by typing new address in serial monitor.

*/

#include <PZEM004Tv30.h>

#define PZEM_RX_PIN 35
#define PZEM_TX_PIN 19

PZEM004Tv30 pzem(Serial1, PZEM_RX_PIN, PZEM_TX_PIN);

/*******************************************
 * Set your address over here..
 * The address can be between 0x01 and 0xF7
 *******************************************/

#define SET_ADDRESS 0x04

String reading;

void setup() {
  Serial.begin(115200);
}

void loop() {
    static uint8_t addr = SET_ADDRESS;

    // Print out current custom address
    Serial.print("Previous address:   0x");
    Serial.println(pzem.readAddress(), HEX);

    Serial.println("enter new address: ( 1 - 254 )");     //Prompt User for
input
    while (Serial.available()==0) {}            //Wait for user input
    reading = Serial.readString();                 //Read user input and
hold it in a variable
    int intread = reading.toInt();
        addr = intread;

    // Set the custom address
    Serial.print("Setting address to: 0x");
    Serial.println(addr, HEX);
    if(!pzem.setAddress(addr))
    {
      // Setting custom address failed. Probably no PZEM connected
      Serial.println("Error setting address.");
    } else {
[... 137 lines cut ...]
    delay(1000);
}

I have 6 pzem on the same uart and it's ok.

Thank you for your job

Michel DUFRESNE
~~~~~~~~~~~~~~~

Le mar. 28 mai 2024 à 13:33, Jakub Mandula ***@***.***> a
écrit :

> When you assign the addresses, are you making sure to only have the one
> PZEM connected on the bus and attached to the AC power supply at a time?
>
> —
> Reply to this email directly, view it on GitHub
> <https://github.com/mandulaj/PZEM-004T-v30/issues/124#issuecomment-2134994262>,
> or unsubscribe
> <https://github.com/notifications/unsubscribe-auth/ALPSWQJZI3RCBD2E3MQZXETZERTRLAVCNFSM6AAAAABH6WZQK6VHI2DSMVQWIX3LMV43OSLTON2WKQ3PNVWWK3TUHMZDCMZUHE4TIMRWGI>
> .
> You are receiving this because you authored the thread.Message ID:
> ***@***.***>
>

mandulaj, 2024-05-29:
So does the setting of the address above `4` work now? What is the exact issue? If it has been solved could you please close the issue?

closed 2024-05-29 by micheldufresne