mandulaj/PZEM-004T-v30 #126: Sometimes the data is not fetched from few of the pzems

frozenmafia, 2024-06-07:
**Cannot consistently read the values**

**To Reproduce**
Steps to reproduce the behavior:
1. Connect 5 pzem on esp32 on rx tx (16 17)
2. change the address of all pzems ( I did 50,51,52,53,54)
3. Now us pzem multi library to fetch data , I am also dispalying the same on lcd (16x2)
4. Some times I see values are not read ( I fetch the data on the interval of 2 sec)

**Expected behavior**
There shall be no miss in data

**Additional context**
Is this normal behaviour, as the values are missed 1 out of 6 times

mandulaj, 2024-06-10:
Having 6 PZEMs on one line can sometimes be a bit demanding on the source pin of your RX/TX GPIO. Make sure the GPIO is able to source enough current to supply all the PZEM opto couplers. Also make sure the wire are not too long.

frozenmafia, 2024-06-10:
> Having 6 PZEMs on one line can sometimes be a bit demanding on the source pin of your RX/TX GPIO. Make sure the GPIO is able to source enough current to supply all the PZEM opto couplers. Also make sure the wire are not too long.

Thanks for the update, Is their any diagram that I can refer ? 
And my wire are not long , max 15 cm long

frozenmafia, 2024-06-10:
I am working on a project that involves using multiple PZEM-004T v3.0 modules to monitor AC power. The modules are connected as shown in the schematic below:
[image]

When I power on the AC supply, I can see readings for about 20 seconds before they all turn to nan. This issue does not occur when only one or two modules are connected. Below is an image of my actual setup:

[image]

**Observations**
My ESP32 is powered from the USB.
From the datasheet, the PZEM-004T has a TTL interface for communication and uses a UART to TTL communication interface with a baud rate of 9600, 8 bits, 1 stop bit, and no parity.
When the readings turn to nan, the RX/TX LEDs blink intermittently.
I suspect that this issue could be related to an inadequate power supply or communication collisions between the modules.
I have read that adding a diode or making changes to the hardware might help resolve this issue.
Code
Here is the main part of my code where I take readings: [PZEMMonitor.cpp](https://github.com/frozenmafia/Terant/blob/cd06471d8de7eb913a2260ca48f9a1a18960992a/src/PZEMMonitor.cpp#L62)

And this is the main loop: [main.cpp](https://github.com/frozenmafia/Terant/blob/cd06471d8de7eb913a2260ca48f9a1a18960992a/src/main.cpp#L44C1-L44C43)

**Request for Assistance**
**Power Supply:**

How can I ensure that the power supply is adequate for all connected modules?
What are the steps to increase the power supply in the GPIO?
**Serial Communication:**

Could the readings of all PZEM modules be colliding, and how can I prevent this?
Is there a specific method to handle multiple PZEM modules to avoid data collision?
**Hardware Adjustments:**

What are the best practices for adding diodes or resistors to stabilize the hardware setup?
Could you provide links to articles or videos that explain these hardware modifications?
Any guidance on how to resolve these issues would be greatly appreciated.
This is the [link to my schema](https://www.canva.com/design/DAGHw7ftBBE/cOdbwVT7u-nLQcjlJUT7UQ/edit?utm_content=DAGHw7ftBBE&utm_campaign=designshare&utm_medium=link2&utm_source=sharebutton) if would be great if you could pin point any mistake.

vortigont, 2024-06-11:
@frozenmafia check [this](https://github.com/vortigont/pzem-edl/blob/main/docs/hwmod.md)

frozenmafia, 2024-06-11:
> @frozenmafia check [this](https://github.com/vortigont/pzem-edl/blob/main/docs/hwmod.md)

I have gone through the link . What I could gather was to remove r4 and replace r8 with 470 ohm resistance I did the same . 
But I still get the same error . Is there a robust fix for this ? [image]

Above is the image of the changes I made to module . Please pin point any thing I did wrong .
Thanking you.

mandulaj, 2024-06-11:
Possible workaround: Since you are using an ESP32 which has extra Serial interfaces, you could split the PZEMs between different once. Putting 2 on one interface, 3 on the other....

frozenmafia, 2024-06-11:
I removed one module , now I have four, these are working better,  for some time I get nan, but after that it shows the reading. I think It confirms that the problem was with the current. I will try by providing external source.

vortigont, 2024-06-12:
you can try installing transistor to RX pins like [here](https://github.com/olehs/PZEM004T/wiki/Connecting-PZEM-to-ESP8266)

mandulaj, 2024-06-12:
Yes, either you use a transistor as @vortigont suggests, or you split to two Serial interfaces

frozenmafia, 2024-06-12:
@vortigont @mandulaj Thank you both for your support. I’ve ordered more modules and will conduct tests on them soon. I’ll be sure to keep you updated.

mandulaj, 2024-08-08:
Hey @frozenmafia, is ther any update on this issue? Otherwise I will close it.