mandulaj/PZEM-004T-v30 #127: CTs of higher rating

shayankhawar, 2024-06-26:
Can a current transformer (CT) with different ratings be used by applying a multiplying factor or modifying the board shunt, such as R18? Can we use another CT apart from the one provided, such as a 100A input and a 100mA output, by using a high-power shunt across the CT terminals to approximate the actual value with a multiplying factor or another method?

For instance, can we use it as a 10A variant by connecting the CT in series instead of a load? Typically, standard industrial CTs have a 5A output. The issue with the 10A variant is that it takes a 220V measurement from the same line as the shunt, so I want to modify the 100A v3 variant a bit. If anyone can help, please let me know.

mandulaj, 2024-06-27:
In theory yes, I would calibrate the PZEM using a known good multimeter as the scaling factor could be something unexpected.

shayankhawar, 2024-06-27:
By calibrating pzem do you mean calculating the value of new shunt as well? Which will replace R18?