mandulaj/PZEM-004T-v30 #135: Multiple pzem-004t, only one communicates when together, but work fine separately

TheodorAraps, 2024-12-08:
Hello, and thanks for this great library. I am trying to create a 3 phase meter using arduino leonardo based MCU and pzem-004t V3.0. For now i am just trying to make 2 of the pzem004t work in tha same modbus. 
For some reason i only managed to make each one (individually connected) to work fine only through hardware serial, using Serial1 (witch are pins 0 and 1 of leonardo used for Tx and Rx). I managed to also set addresses of 0x01 and 0x02 for the 2 pzems using the change address code in your repo. From your code for multiple pzems i am using the Serial2 version (change it to serial1 for my case) and alsa changed baud rate to 9600 cause it doesn't work with 115200. Also some info about the pzems and the circuit:
One of the pzem004t works only with 3v3 source of arduino and not th 5V which is super weird. It has 348K octocouplers. The other works with both 5V and 3v3 and has 139K octocouplers. So for my implementation i am using 3v3 so that both can work. Moreover, using the code i provide you, when i have only one pzem connected (either one) it works fine and the data gets printed correctly. Bute when i connect the other one, only the pzem with the 139K octocouplers prints results. The other prints "Error reading voltage" (NaN values i guess). So something is wrong with the communication? What could be the problem? 

[image]
[image]

The code i use is the following:

```
#include <PZEM004Tv30.h>

#define PZEM_SERIAL Serial1
#define NUM_PZEMS 2

PZEM004Tv30 pzems[NUM_PZEMS];

void setup() {
    Serial.begin(9600);

    // For each PZEM, initialize it
    for(int i = 0; i < NUM_PZEMS; i++)
    { 

        // Initialize the PZEMs with Hardware Serial1 on the default pins

        pzems[i] = PZEM004Tv30(Serial1, 0x01 + i);
    }
}

void loop() {
    for(int i = 0; i < NUM_PZEMS; i++){
        // Print the Address of the PZEM
        Serial.print("PZEM ");
        Serial.print(i);
        Serial.print(" - Address:");
        Serial.println(pzems[i].getAddress(), HEX);
        Serial.println(pzems[i].readAddress(), HEX);    
        Serial.println("===================");

        // Read the data from the sensor
        float voltage = pzems[i].voltage();
        float current = pzems[i].current();
        float power = pzems[i].power();
        float energy = pzems[i].energy();
        float frequency = pzems[i].frequency();
        float pf = pzems[i].pf();

        // Check if the data is valid
        if(isnan(voltage)){
            Serial.println("Error reading voltage");
        } else if (isnan(current)) {
            Serial.println("Error reading current");
        } else if (isnan(power)) {
            Serial.println("Error reading power");
        } else if (isnan(energy)) {
            Serial.println("Error reading energy");
        } else if (isnan(frequency)) {
            Serial.println("Error reading frequency");
        } else if (isnan(pf)) {
            Serial.println("Error reading power factor");
        } else {
            // Print the values to the Serial console
            Serial.print("Voltage: ");      Serial.print(voltage);      Serial.println("V");
            Serial.print("Current: ");      Serial.print(current);      Serial.println("A");
            Serial.print("Power: ");        Serial.print(power);        Serial.println("W");
            Serial.print("Energy: ");       Serial.print(energy,3);     Serial.println("kWh");
            Serial.print("Frequency: ");    Serial.print(frequency, 1); Serial.println("Hz");
            Serial.print("PF: ");           Serial.println(pf);

        }

        Serial.println("-------------------");
        Serial.println();
        delay(200);
    }

    Serial.println();
    delay(3000);
} 
```

mandulaj, 2024-12-10:
It might be that when you connect them in parallel, the opto-coupler with lower input impedance just steals all the current and doesn't allow the other one to activate properly. This is the rough schematic:
[image]

R8,R10, R4 and R9 should be the same on all of the devices.... If one has lower resistnace, that opto coupler will turn on first and all others on the line won't glow.

Either you replace the resistors in order to match them, or you use transistors on the line to source the current for each PZEM individually or you put the PZEMs on different busses.

TheodorAraps, 2024-12-10:
Thanks for the quick reply! Oh i see i was suspecting something like that. Could diodes also help, like in the image? 

[image]

mandulaj, 2024-12-10:
I don't see how that would help anything... You are not eliminating one of the weaker PZEMs taking ownership of the bus. But try it and let me know if it works. :smile:

TheodorAraps, 2024-12-11:
As you expected, it didn't work 😅 waiting for transistors to arrive to try this approach. I also ordered more pzems from the same store and hope they have the same resistance...

TheodorAraps, 2024-12-13:
I didn't succeed with the transistors. I used 2N2222A NPN (0.8A) transistors and the circuit i made is the following:
[image]
Did i do something wrong?