mandulaj/PZEM-004T-v30 #136: Solutions for Reading Data from Three PZEM Devices with ESP32

anassafar1985, 2025-01-22:
Thank you to Mandulaj and their team for their excellent efforts in creating this energy meter. I am working on a program to read data from a three-phase energy meter using an ESP32, but I have encountered several challenges.

Initially, I attempted to use the HardwareSerial method, as the ESP32 has two serial ports. However, I could only obtain results from two PZEM devices, which caused my first attempt to fail. Undeterred, I explored the software serial port method, but it proved incompatible with the ESP32 board.

Next, I tried connecting three PZEM devices in series to a single RX/TX pin, but I still couldn't achieve the correct output. Below are the relevant sections of my code for these attempts:

`//HardwareSerial Method:

#include <PZEM004Tv30.h>
#include "HardwareSerial.h"

HardwareSerial Serial3(1); 
HardwareSerial Serial4(2);
HardwareSerial Serial5(3);

PZEM004Tv30 pzem1(&Serial3, 25, 26); 
PZEM004Tv30 pzem2(&Serial4, 18, 19);
PZEM004Tv30 pzem3(&Serial5, 16, 17);

float VOLTAGE1, CURRENT1, POWER1;
float VOLTAGE2, CURRENT2, POWER2;
float VOLTAGE3, CURRENT3, POWER3;

void setup() 
{
  Serial.begin(115200);
  Serial.println("\nFZEM004T Testing");
  pzem1.setAddress(0x10);
  pzem2.setAddress(0x08);
  pzem3.setAddress(0x09);
}

void loop() 
{
  VOLTAGE1 = pzem1.voltage();
  VOLTAGE1 = zeroIfNan(VOLTAGE1);
  CURRENT1 = pzem1.current();
  CURRENT1 = zeroIfNan(CURRENT1);
  POWER1 = pzem1.power();
  POWER1 = zeroIfNan(POWER1);
  delay(1000);
  Serial.println("");
  Serial.print("Custom Address:");
  Serial.println(pzem1.readAddress(), HEX);
  Serial.printf("Voltage1        : %.2f\ V\n", VOLTAGE1);
  Serial.printf("Current1        : %.2f\ A\n", CURRENT1);
  Serial.printf("Power Active1   : %.2f\ W\n", POWER1);
  Serial.printf("---------- END ----------");
  Serial.println("");

  // same for pzem2 and pzem3
}

float zeroIfNan(float v) 
{
  if (isnan(v)) 
  v = 0;
  return v;
}
`
output
------
Custom Address:30
Voltage1        : 235.10 V
Current1        : 0.00 A
Power Active1   : 0.00 W
---------- END ----------

Custom Address:20
Voltage2        : 235.30 V
Current2        : 0.00 A
Power Active2   : 0.00 W
---------- END ----------

Custom Address:0
Voltage3        : 0.00 V
Current3        : 0.00 A
Power Active3   : 0.00 W
---------- END ----------

`//Series Connection Method:

#include <PZEM004Tv30.h>

PZEM004Tv30 pzem1(&Serial2, 16, 17); 
PZEM004Tv30 pzem2(&Serial2, 16, 17); 
PZEM004Tv30 pzem3(&Serial2, 16, 17);  

float VOLTAGE1, CURRENT1, POWER1;
float VOLTAGE2, CURRENT2, POWER2;
float VOLTAGE3, CURRENT3, POWER3;

void setup() 
{
  Serial.begin(115200); 
  Serial.println("\nPZEM004T Testing");

  Serial2.begin(9600, SERIAL_8N1, 16, 17); 
    pzem1.setAddress(0x10);
    pzem2.setAddress(0x20);
    pzem3.setAddress(0x30);
}

void loop() 
{
  VOLTAGE1 = pzem1.voltage();
  VOLTAGE1 = zeroIfNan(VOLTAGE1);
  CURRENT1 = pzem1.current();
  CURRENT1 = zeroIfNan(CURRENT1);
  POWER1 = pzem1.power();
  POWER1 = zeroIfNan(POWER1);
  Serial.println("\nPZEM1:");
  Serial.print("Custom Address:");
  Serial.println(pzem1.readAddress(), HEX);
  Serial.printf("Voltage1        : %.2f V\n", VOLTAGE1);
  Serial.printf("Current1        : %.2f A\n", CURRENT1);
  Serial.printf("Power Active1   : %.2f W\n", POWER1);
  Serial.println("---------- END ----------");
  delay(1000);

// same fr pzem2 and pzem3
}

float zeroIfNan(float v) 
{
  if (isnan(v)) 
    v = 0;
  return v;
}`

OUTPUT
------

PZEM1:
Custom Address:0
Voltage1        : 0.00 V
Current1        : 0.00 A
Power Active1   : 0.00 W
---------- END ----------

PZEM2:
Custom Address:0
Voltage2        : 0.00 V
Current2        : 0.00 A
Power Active2   : 0.00 W
---------- END ----------

PZEM3:
Custom Address:0
Voltage3        : 0.00 V
Current3        : 0.00 A
Power Active3   : 0.00 W
---------- END ----------

Could you please help me determine whether the **ESP32 can support reading data from three PZEM devices simultaneously**? Any guidance or alternative solutions would be greatly appreciated.

vortigont, 2025-01-25:
you can check [this](https://github.com/vortigont/pzem-edl/blob/main/docs/hwmod.md) and [this](https://github.com/vortigont/pzem-edl) for multiple devices connectivity

anassafar1985, 2025-01-25:
Thank you! I’ve achieved the correct result.

closed 2025-01-25 by anassafar1985

WaeiZ, 2026-03-28:
anassafar1985, can we talk?

I'm facing the same problem.