mandulaj/PZEM-004T-v30 #139: Compatibility with PZEM-004T v4.0

npbk21, 2025-05-03:
Does your library work with the new PZEM-004T v4.0 version?

MrJoiny, 2025-05-06:
It does work. I think V3 and V4 is the same in terms of Modbus communication.

barman75, 2025-05-28:
> [@barman75](https://github.com/barman75) You could try using a PNP transistor (common collector) in your circuit. It worked for me, when I connected more than two PZEM V4.0s. Since the PZEM dont have enough power.

Thanks, I`ll try. I am using 5V actually, so it is strange that power is not enough. I`ve  asked the manufacturer as well but have no response yet.

barman75, 2025-06-13:
> [@barman75](https://github.com/barman75) You could try using a PNP transistor (common collector) in your circuit. It worked for me, when I connected more than two PZEM V4.0s. Since the PZEM dont have enough power.
> 
> [image]

Solved with simple zener diodes